CougsInSpace/CougSat1-Hardware #20: Research Thermal Expansion of PCBs

WattsUp, 2018-04-22:


WattsUp, 2018-07-06:
FR4 vs the copper vs the solder vs the components
Large components could have the leads torn/stressed off with differential expansion

mvguerrero, 2018-10-24:
(The research gathered from the 21st is included in the new slack channel "ec-research" and is available for any to observe)
In conclusion, the heating and cooling of the satellite will fatigue the components of the PCB's. If we want to determine more specific information about parts we are worried about, we need to find the exact material properties of each PCB element , and the exact physical arrangement of the components in question. 
Once we create a baseline for temperature ranges the interior of the satellite reaches we can find the stress developed from differences of thermal expansion. Once our equations of interior temperatures are established we can better predict the extent that this phenomenon would be a problem with our current arrangements of PCB's
 An issue for next sprint would be to use this research to determine which components are at the most risk, and determine from their physical arrangement if the part experiences forces outside the fatigue limit, and if not or how many cycles it would take to reach that limit and fail.

closed 2018-10-24 by mvguerrero