CougsInSpace/CougSat1-Hardware #94: THERMAL - Bakeout Testing With Respect to Payload (due June 16th)

colinwarn, 2018-05-30:
We are currently planning to germinate either corn, wheat or peas:  Each of which can survive between 0 and 38 degrees Celsius optimally.  Given our satellite has to go through thermal vacuum testing, we need thermal to check to see if this testing could potentially kill out seeds: And if so whether we can get a waiver or how we can make sure our seeds survive said testing.

mvguerrero, 2018-08-02:
[627972main_LSP-REQ-317_01A.pdf](https://github.com/CougsInSpace/CougSat1-Hardware/files/2251821/627972main_LSP-REQ-317_01A.pdf)
 Kevin found this about Bakeout Testing

daelinmalizia, 2018-08-02:
We also have a CAD thermal simulation currently in progress. We had some software issues, but now we need to update it to account for radiation.

mvguerrero, 2018-08-28:
We are encountering an issue where as we add more simulated insulation, the internal equilibrium temperature would rise compared to having none. We are unsure if this phenomenon is based in reality or an issue with the simulation.

mvguerrero, 2018-09-02:
This is information from Levi: 
Currently the thermal study is defined as follows, with "bonding" contact sets between the Kapton layers, as well as radiative surfaces defined. Currently the simulation results don't indicate that temperature drops very much. This is probably a fault within the definitions of the simulation and will need to be tried with several differing initial settings as well as try to compare with other similar thermal studies found online. Biggest roadblocks thus far are (1) trying to accurately define the initial settings and (2)  making sure the simulation results are in line with real world experiments.
[image]
[image]
[image]

mvguerrero, 2019-09-15:
MICROTASK: Research Bakeout procedure and  post the conditions of the test. These conditions are used in simulation of the test

mvguerrero, 2019-09-15:
MICROTASK: Use the Grabcad Payload chamber, or its analogue in a thermal simulation which is like the bake out procedure, and list results.

mvguerrero, 2019-09-15:
MICROTASK: Create bakeout test thermal simulation with Bus (satellite body) surface, and list results

mvguerrero, 2019-09-15:
MICROTASK: Combine the bakeout tests together using superposition and list results.

mvguerrero, 2019-09-29:
With the news that NANORACKS does not require a bakeout test to be performed for our hardware to launch. Since Cougsat 1 will ride in a pressurized container alongside ISS supplies, the container will not experience outgassing conditions. Therefore, the extreme temperatures we were concerned about the seeds facing in a bakeout test will no longer be an issue.

closed 2019-09-29 by mvguerrero