CougsInSpace/CougSat1-Hardware #112: Kevin:

mvguerrero, 2018-06-06:
The scholar document for cube sat thermal analysis showed our procedure for our work: 
1: Thermal calculations for cube without internal components
2: Determining orbit time and cube rotation determination
3: Beta angle defining system temperature dependency
4: Power budget and analysis with all components
Difficulty: Hard

Done: Found that cold dependent is a priority, and our beta angle must be less than 60deg.    Still need an exact, part to part power budget
-Kevin

mvguerrero, 2018-06-08:
The scholar document for cube sat thermal analysis showed our procedure for our work:
1: Thermal calculations for cube without internal components
2: Determining orbit time and cube rotation determination
3: Beta angle defining system temperature dependency
4: Power budget and analysis with all components

Done: Found that cold dependent is a priority, and our beta angle must be less than 60deg. Still need an exact, part to part power budget
-Kevin

WattsUp, 2019-09-16:
What is the reason for reopening this?

mvguerrero, 2019-09-25:
Issue was reopened to add more information from the team's old drive. Instead we added the information in the resources tab. Closing issue

closed 2019-09-25 by mvguerrero