CougsInSpace/CougSat1-Hardware #117: COUGSAT-1: Create System Requirements

WattsUp, 2018-06-12:


WattsUp, 2018-07-08:
This also involves a SRR

WattsUp, 2018-07-08:
Need to separate system requirements and design requirements
"The CubeSat shall have contact with the ground station at all times" vs "The RCS should be full duplex"
Must haves vs nice to haves
And design requirement should be traceable to a system requirement
System requirements are a "constitution": able to change but difficult/long process

Change CubeSat specifications to "Must comply with CubeSat Specification 3.1.1" to allow the CubeSat document to change without needing to update our document. It's okay with a small project though. Might add  a revision number that we are referencing to.

Rename priority levels: minimum requirement (non negotiable), can change the value, cosmetic, etc.
Key performance parameters
Bare minimum
Expectations
All design requirements are equal

CodySigvartson, 2018-07-09:
[meetingnotes.docx](https://github.com/CougsInSpace/CougSat1-Hardware/files/2174283/meetingnotes.docx)

This file contains notes on requirement revisions from the SRR. If you have any questions please reach out to me

PR #301 (2018-09-11): System requirements
Fixes #117
closed 2018-09-11 by WattsUp