CougsInSpace/CougSat1-Hardware #243: Frontplane Design

colinwarn, 2018-06-29:
Create a 1-2mm panel for the frontplane of the Cubesat that can integrate:

-Umbilical cord hole
(solar panels?)
(magnetorquer?)
(Thermal insulation?)

WattsUp, 2018-07-02:
It will need pass through for two antenna cables. Maybe three. 
No solar panels nor magnetorquer mounting needed

colinwarn, 2018-07-16:
[Front Plane.PDF](https://github.com/CougsInSpace/CougSat1-Hardware/files/2199541/Front.Plane.PDF)
[Front Plane.zip](https://github.com/CougsInSpace/CougSat1-Hardware/files/2199542/Front.Plane.zip)

Latest update to drawings.

colinwarn, 2018-07-17:
[Front Plane.PDF](https://github.com/CougsInSpace/CougSat1-Hardware/files/2203537/Front.Plane.PDF)
[Front Plane.zip](https://github.com/CougsInSpace/CougSat1-Hardware/files/2203538/Front.Plane.zip)

closed 2018-07-18 by colinwarn