CougsInSpace/CougSat1-Hardware #310: Orbital Design Software

kevvillarreal, 2018-09-09:


TaylanSwift, 2018-09-16:
Update: I found some "Small Sat Tools" outlined by NASA under thermal and radiation analysis. 
Kevin and I got access to Space environment information system (SPENVIS), but it yields obscure graphs and data like electron fluxes, when given orbit altitude, and relevant parameters. Might be useful later on when we know better what "orbital environment" data we need. 
Thermal Desktop was the #1 recc.'d general purpose application - but it' based on AutoCad - which doesn't mesh well with what everyone in CiS is using. 
Then, looking through resources on SolidWorks, it seems that Solid Works Flow Simulation is a decent option for radiation modelling. We'll be investigating how useful it is this week.

mvguerrero, 2018-09-22:
We are meeting to learn how to use NASA's GMAT software to supply important information to our radiation equations. There is also the potential to interface with matlab which we may explore in a later issue.

mvguerrero, 2018-09-22:
Hendrik found out how to export beta angle data from GMAT today, and we all have a basic familiarity of the software. We know how to create orbits, change properties of the satellite, and run orbital simulations.

closed 2018-09-22 by mvguerrero