CougsInSpace/CougSat1-Hardware #311: Radiation Equations

Cory-Palmieri1, 2018-09-09:
Develop a Radiation Equation to simulate our thermal layering.

mvguerrero, 2018-11-11:
After our meeting with Dr. Richards, we learned we are still on the right track. When considering the thermal mass of the insulation, we can consider it as constant because the thickness is so small, thus the temperature difference on either side of the insulation is negligible. However the thermal mass of the satellite such as the frame and the components is not, for now we will be simplifying the satellite into "chunks" of materials for calculations.

closed 2019-03-24 by Mikte

Mikte, 2019-03-24:
Concluded Radiation calculations.
Created two similar equations representing the thermal behavior of the satellite: one equation determining the inside temperature of the satellite and it's components and another determining the exterior temperature of the satellite (mainly MLI with options for more if necessary).

_Determine Temps based on present material of satellite (modifications available)_
[Interior Temps_Alex Lambert.xlsx](https://github.com/CougsInSpace/CougSat1-Hardware/files/3000911/Interior.Temps_Alex.Lambert.xlsx)