CougsInSpace/CougSat1-Hardware #379: ENVIRONMENTAL CONTROL-Research Solar Panel PCB Expansion and Fatigue

mvguerrero, 2018-10-25:
Sunday's research brought a  lot of general info about differential thermal expansion, but we would like to focus on Solar Panel PCB's in this next issue.

mvguerrero, 2018-11-11:
We found the thermal expansion coef for GaAs(5.73*10^-6 celcius^-1) and Germanium (5.9*10^-6 cecius^-1)
We can't find the thermal expansion coef for GALNP2 and we are still searching

jacob-nichols, 2019-04-08:
[image]

I got the PCB thermal expansion coefficients for the gold/copper alloy as well as the fiberglass. The fiberglass coefficient was given in 2 ways, lengthwise and crosswise. I was not able to get information regarding when the difference in the coefficients becomes dangerous.
I used these links to get the information:
Alloy - https://www.engineeringtoolbox.com/linear-expansion-coefficients-d_95.html
FR4 - https://www.farnell.com/datasheets/1644697.pdf
Unit Conversions:
- http://www.unitconversion.org/unit_converter/thermal-expansion.html
- http://asm.matweb.com/search/GetUnits.asp

mvguerrero, 2019-04-14:
Hey, I think we can find the point at which the expansion is dangerous through theory.
Assuming that the danger comes from the strain caused from differential expansion, we need to find the strain developed due to the difference of expansion coefficients. Also we need to know the endurance limit curve for the material with expands slower. The endurance limit is dependent on both the cycles endured, and the strain at which the cycles take place. That would also give us a lifetime or the part. If it's below the curve, then we shouldn't have any trouble, and if it's above, then we need to find the lifetime.

mvguerrero, 2019-11-17:
This situation should not be a problem for our sat, as it will run cold according to our latest calculations. The thermal cycling will be very slight, and the materials involved should not expand enough for us to worry about.

closed 2019-11-17 by mvguerrero