CougsInSpace/CougSat1-Hardware #414: ENVIRONMENTAL CONTROL-Draft Vacuum Chamber Test Procedures

mvguerrero, 2018-11-15:
The goal of this issue is to create a procedure for testing components within the acrylic vacuum chamber with regards to the resistor rig.

mvguerrero, 2019-01-20:
Cody Christiansen @ctchris1 has created a fantastic template for testing procedures. Check it out!
[Test Plan - Template.docx](https://github.com/CougsInSpace/CougSat1-Hardware/files/2777136/Test.Plan.-.Template.docx)

mvguerrero, 2019-06-03:
Levi did some even more fantastic work on drafting the procedures from the plans of @ctchris1 .
Attached is the file. 
[Uploading Draft_Insulation_Installation_REV01 (1).docx…]()

closed 2019-06-03 by mvguerrero

LeviJCarroll, 2019-09-15:
[Draft_Insulation_Installation_REV01.1.docx](https://github.com/CougsInSpace/CougSat1-Hardware/files/3614330/Draft_Insulation_Installation_REV01.1.docx)

@mvguerrero Document change for the insulation installation (Revision 1.1).
Changes include:
-Changing authority and relevant personnel to reflect moved members
-Added Daelin as a safety contact
-Minor formatting changes