CougsInSpace/CougSat1-Hardware #470: ENVIRONMENTAL CONTROL - Draft Procedure for MLI Blanket

LeviJCarroll, 2019-04-01:
Draft procedure that outlines how to construct the MLI blanket needed to apply to Coug-Sat 1.

LeviJCarroll, 2019-04-01:
[Draft_MLI_PROCEDURE.docx](https://github.com/CougsInSpace/CougSat1-Hardware/files/3027416/Draft_MLI_PROCEDURE.docx)

@mvguerrero If you can take a look at this to make sure it's satisfactory

LeviCarroll, 2019-04-07:
[MLI Procedure_Rev01.docx](https://github.com/CougsInSpace/CougSat1-Hardware/files/3052095/MLI.Procedure_Rev01.docx)

Final draft of MLI Document for revision 01

mvguerrero, 2019-04-07:
The above MLI assembly procedure is an important milestone on the team's way to providing proper documentation to its future members. This document sets and excellent example of the level of detail expected of such procedures, and stores this knowledge for future members to readily use.  I am extremely impressed with Levi's work, and the scrutiny provided by Kevin and Alex.
-Mario Guerrero

closed 2019-04-07 by mvguerrero

LeviJCarroll, 2019-09-15:
[MLI Procedure_Rev01.1.docx](https://github.com/CougsInSpace/CougSat1-Hardware/files/3614300/MLI.Procedure_Rev01.1.docx)

@mvguerrero Here's the updated version of the MLI document (revision 1.1). Changed include:
-Changing authority and relevant personnel to reflect graduated members
-Added Daelin as a safety contact
-Minor formatting changed