CougsInSpace/CougSat1-Hardware #535: ENVIRONMENTAL CONTROL - Vacuum Chamber Testing Procedure Radiation

LeviJCarroll, 2019-09-15:
Draft official procedure for using using the vacuum chamber for the radiation simulation. This will be using an electrical circuit configuration including resistors, wires, and a power source to simulate the 3W power that the satellite will experience in LEO.

Major subjects include:
- Safe handling of chamber including startup and shutdown
- Satellite frame with MLI layer to see effectiveness of blanket
- Circuit configuration for power output

mvguerrero, 2019-09-15:
MICROTASK: Update information to include the new safety officer, Dalin

LeviJCarroll, 2019-09-15:
> 
> 
> MICROTASK: Update information to include the new safety officer, Dalin

It will be added in the first draft of the document; it hasn't been created yet

mvguerrero, 2019-11-17:
Levi is working hard on the final design for the chamber, and intends to post about the goals for the documentation of the device, as it comes together.