CougsInSpace/CougSat1-Hardware #539: ENVIRONMENTAL CONTROL: Review Orbital Equations

mvguerrero, 2019-09-22:
Kevin is here to explain the Oribital Temperature Equations he had made in the past. We are taking notes and pictures of his explanations to further bolster our understanding of how LEO affects our surface temperature.

mvguerrero, 2019-09-22:
Here is the whiteboard writing that Kevin made
[image]

kevvillarreal, 2019-09-23:
These are the complete calculations used. These come directly from the SMAD.
Because our cube is so small we can assume that the internal temperature will be the same as the external temperature.
Each photo is numbered according to page order.

NOTE: Because the heat flux equations (Qsolar, Qalbedo, etc.) take into account the beta angle inside the cosine, the values are smaller for the flux as beta reaches 45 degrees.

[Calculating Temperature.zip](https://github.com/CougsInSpace/CougSat1-Hardware/files/3640434/Calculating.Temperature.zip)

mvguerrero, 2019-09-26:
This review session was very productive, and allowed new members to understand the existing sheet, and how we can improve its accuracy in the near future. Soon we will modify the sheet to include the anodized surface in our calculations.

closed 2019-09-26 by mvguerrero