CougsInSpace/CougSat1-Hardware #540: ENVIRONMENTAL CONTROL: Research Water Expansion from Freezing

mvguerrero, 2019-09-22:
The water within the payload chamber could freeze between the satilite powering and deployment. We need to know to what factor the water will increase in volume when frozen. The payload team will be able to know how much to account for. The known volume of water which they plan for is 5ml or water. How much more space does it take up when frozen?

mvguerrero, 2019-09-22:
Carson is planning to work on this issue, but he needs to be added to the GitHub first. This comment will be deleted when he does.

mvguerrero, 2019-09-26:
Carson collected a lot of information about his subject. To summarize, the expansion of water should only increase its volume by around 9 percent. For a volume as small as 5ml, the expansion is not very significant. Carson also calculated the energy required to bring 5ml of frozen water back to liquid. Again, this quantity of energy is relatively small, and within power budget. Carson will attach information to this post in future

closed 2019-09-26 by mvguerrero

carson-mock, 2019-09-29:
[Water Thermal Properties.docx](https://github.com/CougsInSpace/CougSat1-Hardware/files/3668170/Water.Thermal.Properties.docx)