CougsInSpace/CougSat1-Hardware #549: ENVIRONMENTAL CONTROL- Calculate the difference between an Anodized dark grey bus, and non-Anodized bus.

mvguerrero, 2019-09-29:
Since the aluminum's surface has changed from our previous calculations, we must determine the impact that this decision could have. Simulate a completely annodized aluminum surface to represent an extreme scenario, and compare that to the more reasonable ratio of solar pannels and annodized surface.

Include the previous results which used a shiny aluminum surface as well.

jacob-nichols, 2019-10-13:
With the current calculations:
     - Cold Case Temperature: -10.7 C
     - Hot Case Temperature: -4.4 C

mvguerrero, 2019-10-27:
[Temperature Analysis Update October.xlsx](https://github.com/CougsInSpace/CougSat1-Hardware/files/3776454/Temperature.Analysis.Update.October.xlsx)
This is Jacob's work for the calculations. Other Contributors researched the governing effects, and talked to teams about materials.

closed 2019-10-27 by mvguerrero