CougsInSpace/CougSat1-Hardware #556: ENVIRONMENTAL CONTROL- Prototype Cooling Chamber Components.

mvguerrero, 2019-10-13:
Use Dry Ice and metal wires to conduct a study about how well metal wires can remove heat from a closed system. This method is currently the main way to remove heat from a system for testing purposes, and its efficacy should be tested in the real world. 

The current conduction candidates are copper and aluminum. Copper has greater conductance than aluminum, but aluminum is much less expensive.

pteridophobia, 2019-10-20:
We may need to wrap the exposed copper with an insulator to allow the other cool to cool

mvguerrero, 2019-10-27:
The experiment informed us that aluminum would not be the ideal heat conductor for the chamber, and that coiling copper wire could work to remove heat from a system. We will have to source a medium diameter copper wire compatible with our planned system. Additional research should be done to find the exact diameter.
[image]

closed 2019-10-27 by mvguerrero